ros-tooling/generate_parameter_library #56: Examples showing proper use within a Component library

spiderkeys, 2022-08-23:
Are there any examples of using this within a component library in a public repo? 

The usage within code is straightforward in this case, but I am trying to figure out:
1.) what a good project structure would look like when creating a publicly consumable Component package, and 
2.) what steps need to be taken in CMake to properly expose dependencies for both static and runtime composition in this scenario

tylerjw, 2022-08-23:
I can try to put something together quickly for this if you'd like or you are welcome to open a PR to do that if you'd like?

spiderkeys, 2022-08-23:
If you have the time, it would be helpful if you could put a simple example together (no rush). I'm not sure exactly how to best do this just yet, which prompted the question.

tylerjw, 2022-08-23:
> Originally, I had seen this issue in MoveIt and was wondering if there might be some relevant WIP that demonstrated this:
https://github.com/ros-planning/moveit2/issues/1455

Working on that task has led me down the rabbit hole that prompted this post: https://github.com/ros-planning/moveit2/discussions/1517

I am currently working on implementing an IK solver that tries to adhere to these principles better. Once I finish that I will resume looking into the interior parts of MoveIt and trying to make it more composable and useful.

tylerjw, 2022-09-06:
I created a PR that changes the example to be a component node. Do you mind reviewing my PR and letting me know if that was what you were looking for?

closed 2022-09-07 by tylerjw

spiderkeys, 2022-09-07:
Hi @tylerjw, sorry just now seeing this. Yes, the PR looked good to me and covered what I was interested in. Appreciate the effort!

PR #60 (2022-09-07): Example component node usage
Fixes #56 

This also fixes some deadlock conditions that were possible and updates the example readme.