ros-tooling/generate_parameter_library #70: Compile error under debian bullseye

mosfet80, 2022-09-21:
<pre>--- stderr: generate_parameter_library_example
<font color="#C01C28">CMake Error at /home/r/ws_moveit2/install/generate_parameter_library/share/generate_parameter_library/cmake/generate_parameter_library.cmake:33 (message):</font>
<font color="#C01C28">  generate_parameter_library_py() variable</font>
<font color="#C01C28">  &apos;generate_parameter_library_py_BIN&apos; must not be empty</font>
<font color="#C01C28">Call Stack (most recent call first):</font>
<font color="#C01C28">  CMakeLists.txt:9 (generate_parameter_library)</font>

</pre>

tylerjw, 2022-09-21:
This means that it is not finding the `generate_parameter_library_py` python app that does the code generation. Can you help me understand how you are building or using generate_parameter_library better so I can understand how to reproduce the error you are seeing?

What commands do you enter when you built this? What other related packages are in your workspace?

--

Also, just so you know I removed `tcb_span` from this repo and am in the process of releasing it from a different repo. If you are building the main branch of this repo you'll need tcb_span from either the ros buildfarm or by putting this other repo in your workspace: https://github.com/PickNikRobotics/cpp_polyfills

Once the current release is synced this problem should go away.

mosfet80, 2022-09-22:
I successfully compiled and tested ros humble on debian (using modified rosdep https://github.com/mosfet80/rosdistro.git)  .. .
The error occurs when I try to compile moveit2 (I changed the repository sources because the compiled version on debian does not exist. https://github.com/mosfet80/moveit2.git) .  i have manual add urdfom_headers and urdfom  to src.

when i build using colcon build --event-handlers desktop_notification- status- --cmake-args -DCMAKE_BUILD_TYPE=Release --os=debian:bullseye

when compile it fails with the error I sent

i tried to use a previus generate_parameter_library version  that tcb_span modification ..with the same result.

tylerjw, 2022-09-22:
So you know this is the line that is reporting that error: https://github.com/PickNikRobotics/generate_parameter_library/blob/7e8a89f0002f30326616d8d0963961bbbdec04b9/generate_parameter_library/cmake/generate_parameter_library.cmake#L31

I copied this pattern from packages from the upstream ros ecosystem where you need to shell out and call a python executable from cmake.

mosfet80, 2022-11-07:
The problem wasn't debian.
I had downloaded the source in the src/PickNikRobotics/generate_parameter_library folder

it was enough to move the code to the src/generate_parameter_library folder to compile without errors.

closed 2022-11-07 by mosfet80