ros-tooling/generate_parameter_library #72: CMake 3.22 requirement prevents build on ROS 2 Humble recommended platforms

nuclearsandwich, 2022-09-27:
The packages `generate_parameter_library` and `parameter_traits` in this repository declare a requirement on CMake 3.22.
While this version is available on the required platforms (Ubuntu 22.04 and Windows 10) listed in [REP-2000](https://www.ros.org/reps/rep-2000.html#humble-hawksbill-may-2022-may-2027), it is not available for the recommended platforms. In particular RHEL 8 which we build binary packages for by default on the build farm.

If it's possible to relax this CMake version constraint to CMake 3.16 it would add support for Ubuntu Focal, Debian Bullseye, and RHEL 8. If not, I'll go ahead and block this package from building on RHEL 8.

tylerjw, 2022-09-27:
Thank you for the issue! I'll work on verifying it works on older cmake. I think I should be able to work around having older cmake.

closed 2022-09-27 by tylerjw

tylerjw, 2022-09-28:
@nuclearsandwich here are PRs to rosdistro that release this and its one dependency with cmake minimum reduced to 3.16. It turns out I was using no features of newer cmake and it just works on older cmake:

* https://github.com/ros/rosdistro/pull/34681
* https://github.com/ros/rosdistro/pull/34682
* https://github.com/ros/rosdistro/pull/34692
* https://github.com/ros/rosdistro/pull/34693

Thank you for opening this issue. Hopefully these PRs fix the issue for RHEL 8.

nuclearsandwich, 2022-09-28:
Thanks very much for the swift response. I've merged the Rolling releases in order to try and resolve these packages for the [pending sync](https://discourse.ros.org/t/preparing-for-rolling-sync-2022-09-29/27533/2).

PR #73 (2022-09-27): Drop requirement for CMake to 3.16
Fixes #72