ros-tooling/generate_parameter_library #85: Cannot build generate_parameter_library_example on RHEL 8

jekorhon, 2022-10-31:
Hi,

We are trying to build MoveIT2 on RHEL 8 and we got problems with this generate_parameter_library. Can you help us to solve it and what is the minimun required Python version to build this? Our error message is following:

```--- stderr: generate_parameter_library_example
Traceback (most recent call last):
  File "/wrk1/ws_moveit2/install/generate_parameter_library_py/bin/generate_parameter_library_py", line 33, in <module>
    sys.exit(load_entry_point('generate-parameter-library-py==0.2.7', 'console_scripts', 'generate_parameter_library_py')())
  File "/wrk1/ws_moveit2/install/generate_parameter_library_py/bin/generate_parameter_library_py", line 25, in importlib_load_entry_point
    return next(matches).load()
  File "/usr/local/lib/python3.6/site-packages/importlib_metadata/__init__.py", line 194, in load
    module = import_module(match.group('module'))
  File "/usr/lib64/python3.6/importlib/__init__.py", line 126, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 994, in _gcd_import
  File "<frozen importlib._bootstrap>", line 971, in _find_and_load
  File "<frozen importlib._bootstrap>", line 955, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 665, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 678, in exec_module
  File "<frozen importlib._bootstrap>", line 219, in _call_with_frames_removed
  File "/wrk1/ws_moveit2/install/generate_parameter_library_py/lib/python3.6/site-packages/generate_parameter_library_py/main.py", line 419, in <module>
    class DeclareStruct:
  File "/wrk1/ws_moveit2/install/generate_parameter_library_py/lib/python3.6/site-packages/generate_parameter_library_py/main.py", line 421, in DeclareStruct
    def __init__(self, struct_name: str, fields: list[VariableDeclaration]):
TypeError: 'type' object is not subscriptable
gmake[2]: *** [CMakeFiles/admittance_controller_parameters.dir/build.make:75: admittance_controller_parameters/include/admittance_controller_parameters.hpp] Error 1
gmake[1]: *** [CMakeFiles/Makefile2:181: CMakeFiles/admittance_controller_parameters.dir/all] Error 2
gmake: *** [Makefile:146: all] Error 2
---
```

Command to build is: `PKG_CONFIG_PATH=/usr/local/lib/pkgconfig:/usr/local/lib64/pkgconfig colcon build --event-handlers desktop_notification- status- --cmake-args -DCMAKE_BUILD_TYPE=Release`

tylerjw, 2022-11-02:
I am not sure how to test what the minimum python version is. We developed this on Ubuntu 22.04, which has Python 3.10.6. Do you know of a good way to test what the minimum python version is?

cottsay, 2022-11-03:
The minimum Python 3 version for all of Galactic, Humble, and Rolling is currently 3.6.8 according to [REP 2000](https://www.ros.org/reps/rep-2000.html#humble-hawksbill-may-2022-may-2027).

I don't have any great suggestions for how to automate testing under Python 3.6 right now, particularly under Ubuntu.

cottsay, 2022-11-03:
Here is an official package build which is broken by this issue: https://build.ros2.org/view/Rbin_rhel_el864/job/Rbin_rhel_el864__joint_state_broadcaster__rhel_8_x86_64__binary/207/

cottsay, 2022-11-03:
Even after resolving this issue, it appears that this package currently requires relatively recent additions to PyYAML:
```
Traceback (most recent call last):
  File "/opt/ros/rolling/bin/generate_parameter_library_py", line 11, in <module>
    load_entry_point('generate-parameter-library-py==0.2.7', 'console_scripts', 'generate_parameter_library_py')()
  File "/opt/ros/rolling/lib/python3.6/site-packages/generate_parameter_library_py/main.py", line 1014, in main
    gen_param_struct.run()
  File "/opt/ros/rolling/lib/python3.6/site-packages/generate_parameter_library_py/main.py", line 990, in run
    docs = yaml.load_all(f, Loader=yaml.FullLoader)
AttributeError: module 'yaml' has no attribute 'FullLoader'
```

tylerjw, 2022-11-03:
I merged the two PRs @cottsay opened. Is there some way you can test that these changes are enough to fix this issue so I can cut a new release and resolve the buildfarm failures?

closed 2022-11-03 by tylerjw

cottsay, 2022-11-03:
> Is there some way you can test that these changes are enough to fix this issue

Both issues that I observed in `joint_state_broadcaster` were reproduced by building `generate_parameter_library_example` as well. I confirmed that both packages build successfully after the two merged fixes were applied, so as far as I know, a release should unblock things.

Thanks for your quick action here.

tylerjw, 2022-11-03:
Thank you for figuring out how to fix these issues!

Cut a patch released and bloomed:
- https://github.com/ros/rosdistro/pull/35186
- https://github.com/ros/rosdistro/pull/35187

PR #88 (2022-11-03): Use YAML loader which is compatible with PyYAML 3.12
RHEL 8 packages PyYAML 3.12, which does not have the `FullLoader` referenced here. If the `SafeLoader` can't be used, the only other option available is the `Loader` (aka `UnsafeLoader`).

Fixes #85
PR #87 (2022-11-03): Use typing syntax which is compatible with Python 3.6
Fixes #85