ros-tooling/generate_parameter_library #86: Option to print parameter values

diogoalmeida, 2022-11-02:
I have been using this library and I am really happy with it - thanks for the contribution!

This issue is more of a question/feature request: can we log the parameters that are loaded by the parameter listener? This would be very useful for debugging purposes.

tylerjw, 2022-11-02:
Would it help if we used ROS logging and sent it to a debug level? Then you could enable it if you want to see them.

diogoalmeida, 2022-11-02:
Absolutely @tylerjw , that would be amazing :)

tylerjw, 2022-11-02:
I'm not sure when I'll get to this, but I will add it to the list of things I'll do next on this project. It might be a couple of weeks as I'm going on vacation tomorrow.

diogoalmeida, 2022-11-02:
thanks for welcoming the feature request, and have a great time off :)

tylerjw, 2023-04-19:
@diogoalmeida I'm sorry it took me so long to get around to this. I just submitted a PR to build this feature. I would appreciate a review if you have time?

diogoalmeida, 2023-04-20:
Hey Tyler, thanks so much for being able to implement this feature. This is exciting :D 

I will definitely give you a review and try it out in our system as soon as possible. 

Once again, many thanks.

PR #106 (2023-04-20): Log when a parameter is set
Fixes #86

If you build this locally, you can test this by running the example node with debug logging:
```shell
ros2 run generate_parameter_library_example test_node --ros-args --params-file src/generate_parameter_library/example/config/implementation.yaml --log-level debug
```

Then in another terminal change a parameter:
```shell
ros2 param set /admittance_controller control.frame.id new_frame
```

You should see an output like this in the logs:
```
[DEBUG] [1681944620.098785049] [admittance_controller]: control.frame.id: string = new_frame
```
closed 2023-04-20 by tylerjw