ros-tooling/generate_parameter_library #94: Could not find a package configuration file provided by "rsl" with any of the following names: rslConfig.cmake,  rsl-config.cmake

pathakarvindk, 2022-11-21:
While building the pkg, getting this issue.

Can anyone help in this regard?
Thank you

-- stderr: parameter_traits                                                                                                                                                     
CMake Error at CMakeLists.txt:8 (find_package):
  By not providing "Findrsl.cmake" in CMAKE_MODULE_PATH this project has
  asked CMake to find a package configuration file provided by "rsl", but
  CMake did not find one.

  Could not find a package configuration file provided by "rsl" with any of
  the following names:

    rslConfig.cmake
    rsl-config.cmake

  Add the installation prefix of "rsl" to CMAKE_PREFIX_PATH or set "rsl_DIR"
  to a directory containing one of the above files.  If "rsl" provides a
  separate development package or SDK, be sure it has been installed.

tylerjw, 2022-11-21:
Assuming you are on Humble, that is because we added the dependency RSL and the release has not synced yet. To fix this locally add RSL to your workspace. Once a sync happens this error should go away.

https://github.com/pickNikRobotics/RSL

If you are on Rolling the sync happened and you should be able to install RSL with rosdep install.

pathakarvindk, 2022-11-23:
@tylerjw, Thanks for your prompt reply.

Do we have any alternative for using the same in Galactic or Foxy.

closed 2022-12-06 by tylerjw