ros-tooling/generate_parameter_library #95: map for dynamic parameters doesn't work

andy-Chien, 2022-12-06:
System:
  Ubuntu 22.04
  ROS Humble
  generate-parameter-library 0.2.8

I can't get the dynamic parameters I setted, only get the default value.
Then I ckecked the header file generated by this lib, found the veriable used to point to the value of dynamic parameters map is not using a reference.

For example, Part of the generated code for setting dynamic params as follows

1    // declare and set all dynamic parameters
2   for (const auto & value : updated_params.name_array){
3        auto entry = updated_params.test.name_array_map[value];
4        auto param_name = fmt::format("{}{}.{}.{}", prefix_, "test", value, "test_1");
5       if (!parameters_interface_->has_parameter(param_name)) {
6            rcl_interfaces::msg::ParameterDescriptor descriptor;
7            descriptor.description = "";
8            descriptor.read_only = false;
9            auto parameter = rclcpp::ParameterValue(entry.test_1);
10          parameters_interface_->declare_parameter(param_name, parameter, descriptor);
11     }
12    param = parameters_interface_->get_parameter(param_name);
13    entry.test_1 = param.as_bool();}

Replace line 3 as below and use the modified header as source code to build, it works.

    auto & entry = updated_params.test.name_array_map[value];

But I don't know how to modify this library to make the generated header correct.
Is it only me having this problem? Or maybe I'm misunderstanding something?

tylerjw, 2022-12-06:
That line is generated from here: https://github.com/PickNikRobotics/generate_parameter_library/blob/ba273649b09f4bf00b55e42a84376037548f19aa/generate_parameter_library_py/generate_parameter_library_py/jinja_templates/declare_runtime_parameter#L3

As you can see it now is `auto& entry = ...`. In the latest version released into Humble (apt update should fix this for you) 0.3.0 this bug is fixed.

```
apt info ros-humble-generate-parameter-library
Version: 0.3.0-1jammy.20221202.034649
```

andy-Chien, 2022-12-07:
Okay got it, thanks!
But when I try to update, the latest version available to upgrade for humble is 0.2.8, is there something I should or I can do to get the new version? or just wait a few days until it's available.

tylerjw, 2022-12-07:
I believe it is available now, you may need to run `rosdep update` and `apt update` then `apt upgrade`. If that doesn't work you can build it from source. If you do that you'll also need the library rsl in your workspace: https://github.com/pickNikRobotics/RSL

andy-Chien, 2022-12-08:
I see, thanks a lot :+1:

closed 2022-12-08 by tylerjw