ros-tooling/generate_parameter_library #97: Custom Unknown Structures under Parameters

destogl, 2023-02-28:
Hi, question / idea about adding unknown complex structures under parameters defined with generate_parameter_library.

## Problem description

I would like to define [this structure of filters](https://github.com/ros-controls/ros2_control_demos/pull/206/files#diff-f86dcfc3f3372de32d4b3c24835485a2ed62957431cb784e926618c337f0a789R30-R52) - see bellow. The purpose of [filter chain](https://github.com/ros/filters/blob/ros2/include/filters/filter_chain.hpp) is to enable flexible setup of arbitrary number of filters that automatically pipe data through it. The filters are plugins and have their own set of parameters, which I am planning to define using generate_parameter_library. But we don't know in advance how many filters there are nor which parameters they have. The filters can take care about their parameters so the outside class (in this case controller) doesn't have to take care about it. 

Target structure:
```
    sensor_filter_chain:
      filter1:
        type: control_filters/LowPassFilterWrench
        name: low_pass_filter
        params:
          sampling_frequency: 200.0
          damping_frequency: 50.0
          damping_intensity: 1.0
          divider: 1

      filter2:
        type: control_filters/GravityCompensationWrench
        name: tool_gravity_compensation
        params:
          CoG:
            x: 0.0
            y: 0.0
            z: 0.1
          force: 0.0        # mass * 9.81
          force_frame: wrist_3_link
          sensor_frame: ft_frame
          world_frame: base_link
```

## How it could be done and what is missing.

The only thing I am missing here is: how can one define “sensor_filter_chain” parameter which is like a “namespace” of parameters under it in such a way that generate_parameters_library accepts that at the outer class/controller level. I could live without generate_parameters_library automatically checking those parameters, it should just “skip” (accept) them for now.

Currently, we can define only parameter type there, but not something “unknown”. Would it make sense or be possible to add something like this:

```
some_controller:
  filter_chain: {
    type: UnknownParametersStructure,
    description: "Map of values that defined filter chain, containing filterN as key and underlying map of parameters needed for a specific filter. See <some docs> for more details.",
  }
```

## Possible workaround (would like to avoid)

Of course, we can simply declare manually this parameter “prefix”/“namespace”, but I would like to avoid that because of mixing two concepts in one node that would be confusing.

saikishor, 2023-03-30:
+1 for this feature. This is a very interesting and useful feature. I would like to isolate my parameters in their own structs and namespaces as they are more meaningful together rather than each of them in individual vectors.

Thank you for bringing it up @destogl

tylerjw, 2023-04-20:
I believe this is now possible with the PR from #99

closed 2023-04-20 by tylerjw