ros-tooling/generate_parameter_library #98: Python Parameter Interface

Flova, 2023-03-08:
Are there any plans regarding a Python integration?

The current handling of parameters in Python under stock ROS2 suffers from similar issues compared to C++.
While Python is dynamic enough to handle generation of a config class during runtime, code generation during build time could improve the experience while autocompleting parameter names in the IDE etc..

I assume integration of CMakeLists.txt based (python) packages is easier compared to setup.py based ones.

tylerjw, 2023-03-08:
I don't regularly use the python interface to ROS so I had no idea it also needed something like this but if you wanted to work on a python code generation output I'd be happy to try to review it and merge your work. We did use jinja templates and python to generate so maybe that would also work for python?

tylerjw, 2023-05-15:
Python interface was just merged. Check it out!

closed 2023-05-15 by tylerjw