ros-tooling/generate_parameter_library #109: Use parameter events?

tylerjw, 2023-04-25:
I should read about and understand the parameter events from ROS and update how generate_parameter_library to use them if appropriate.

fmauch, 2023-05-31:
I would love to see that feature being added to generate_parameter_library :-)

sven-hoek, 2025-09-16:
Not sure if it's possible to register those events inside a generated-parameter-library (as during compilation thereof, the Node isn't known yet). However, the examples could probably simply use the `add_on_set_parameters_callback()` instead of registering a timer callback.

That callback also gets a vector of parameters that did change (which AFAIK is tricky to find out with this library's callback), so can handle such parameter changes accordingly.