ros-tooling/generate_parameter_library #118: Markdown: Nested parameters and __map_

christophfroehlich, 2023-05-18:
I kindly submit two feature requests for the markdown generation 😇

- Nested parameters: one could simply change `self.declare_parameters.code_gen_variable.name` to `self.declare_parameters.parameter_name`, but maybe it can also be exported with this hierarchy to consider that in the jinja2 template?
-  `__map_ ` : these ones are completely ignored now

My understanding of the python/jinja scripts is too low. But if someone points me in the direction I could help to implement these.

tylerjw, 2023-06-15:
@christophfroehlich thank you for the issue. I don't clearly understand your first point. Would you mind breaking it into a separate issue and leaving this about the non-support for map. I don't have time to work on this now but will happily accept PRs that fix this. I'll try to find some time to better document the python/jinja so it is easier to modify for you. I too struggle to edit it efficiently and some better design documentation would be good for me to write because it would force me to finally learn the little details.

christophfroehlich, 2023-06-16:
@tylerjw see #126 for a more detailed explanation of my problem with the nested structures. 
Regarding the map parameters here: Somethink like we have with the[ JTC docs ](https://control.ros.org/master/doc/ros2_controllers/joint_trajectory_controller/doc/userdoc.html)would be good enough:

[image]

PR #164 (2024-01-03): markdown/rst: Support __map_ and nested parameters
This closes #118 and closes #126.

I had to write a custom jinja2 filter `valid_string_cpp` to create valid c++ string literals for multi-line descriptions.

This also adds rudimentary support for the default config:
```yaml
joint_trajectory_controller:
  ros__parameters:
    action_monitor_rate: 20.0
    allow_integration_in_goal_trajectories: false
    allow_nonzero_velocity_at_trajectory_end: false
    allow_partial_joints_goal: false
    cmd_timeout: 0.0
    command_interfaces: '{}'
    command_joints: '{}'
    constraints:
      <joints>: [... cut]
closed 2024-01-03 by pac48