ros-tooling/generate_parameter_library #120: Support using class members/functions to validate parameter

sisaha9, 2023-05-19:
I have a use case where I use a paremeter to read all files in a directory, create a set of valid values and validate the remaining parameters to be one of the values in the set. Is it possible to do this using this library?

tylerjw, 2023-06-15:
The validators are free functions in spirit, however, there are ways you could maybe get around that if you cleverly used some global state. In complex dependent validation situations like this in ros2_control I just wrote extra checks after I read the parameter struct and failed there. It is less useful to the user and less elegant but it is the best work-around I have found so far for this situation where you have dependencies between the values of one parameter and validation of another.

sisaha9, 2023-06-16:
@tylerjw Thanks for the response! Could you share some examples in `ros2_control`?

tylerjw, 2023-06-16:
Here is an example where we do some more advanced checking of the parameters after reading them from ROS in the on_configure method: https://github.com/ros-controls/ros2_controllers/blob/f059022622e12115d006e0677ca2e312eed78858/joint_trajectory_controller/src/joint_trajectory_controller.cpp#L616

sisaha9, 2023-06-16:
Ah I see. I thought this was directly in the validator. Makes sense and it is what I have adopted finally for my use case. Thanks!

closed 2023-06-19 by tylerjw