ros-tooling/generate_parameter_library #124: TypeError: isinstance()

christophfroehlich, 2023-06-05:
I try to compile ros2_controllers within a freshly built ros:rolling docker container, but generate_parameter_library (installed as binary) fails.

> --- stderr: forward_command_controller                                                                                                 
> Traceback (most recent call last):
>   File "/opt/ros/rolling/bin/generate_parameter_library_py", line 33, in <module>
>     sys.exit(load_entry_point('generate-parameter-library-py==0.3.3', 'console_scripts', 'generate_parameter_library_py')())
>   File "/opt/ros/rolling/lib/python3.10/site-packages/generate_parameter_library_py/main.py", line 1035, in main
>     gen_param_struct.run()
>   File "/opt/ros/rolling/lib/python3.10/site-packages/generate_parameter_library_py/main.py", line 1023, in run
>     self.parse_dict(self.namespace, doc[self.namespace], [])
>   File "/opt/ros/rolling/lib/python3.10/site-packages/generate_parameter_library_py/main.py", line 949, in parse_dict
>     self.parse_dict(key, root_map[key], nested_name)
>   File "/opt/ros/rolling/lib/python3.10/site-packages/generate_parameter_library_py/main.py", line 955, in parse_dict
>     self.parse_params(name, root_map, nested_name)
>   File "/opt/ros/rolling/lib/python3.10/site-packages/generate_parameter_library_py/main.py", line 861, in parse_params
>     ) = preprocess_inputs(name, value, nested_name_list)
>   File "/opt/ros/rolling/lib/python3.10/site-packages/generate_parameter_library_py/main.py", line 807, in preprocess_inputs
>     code_gen_variable = CodeGenVariable(
>   File "/opt/ros/rolling/lib/python3.10/site-packages/generate_parameter_library_py/main.py", line 338, in __init__
>     def __init__(
>   File "/usr/local/lib/python3.10/dist-packages/typeguard/_functions.py", line 113, in check_argument_types
>     check_type_internal(value, expected_type, memo=memo)
>   File "/usr/local/lib/python3.10/dist-packages/typeguard/_checkers.py", line 679, in check_type_internal
>     if not isinstance(value, origin_type):
> TypeError: isinstance() arg 2 must be a type, a tuple of types, or a union
> gmake[2]: *** [CMakeFiles/forward_command_controller_parameters.dir/build.make:71: forward_command_controller_parameters/include/forward_command_controller_parameters.hpp] Error 1
> gmake[1]: *** [CMakeFiles/Makefile2:186: CMakeFiles/forward_command_controller_parameters.dir/all] Error 2
> gmake: *** [Makefile:146: all] Error 2

If I try to compile it from source, the generate_parameter_module_example throws the same errors. 
What can be the cause here? 

> Python 3.10.6
> generate-parameter-library-py        0.3.3
> type-description-interfaces          1.7.0
> typeguard                            3.0.2
> typing_extensions                    4.6.3

christophfroehlich, 2023-06-05:
I don't know what happened, after the 5th rebuild of the container it suddenly compiles again.

closed 2023-06-05 by christophfroehlich

sea-bass, 2023-07-07:
I ran into this too, turns out it's because I had some other Python library (`pygls`) that was pinning my `typeguard` version so that 4.0.0 was not installable.

After doing this, my issue was resolved:

```
pip3 uninstall pygls
pip3 install typeguard==4.0.0
```

tylerjw, 2023-07-07:
The day Rust is supported in the buildfarm, the python code in here is gone. I hate leaky python environments.