ros-tooling/generate_parameter_library #125: Use node logger

tylerjw, 2023-06-15:
The logging generated by generate_parameter_library uses namespaced but not node logger. There are downsides to this and we should update to using the node logger as called out here:

https://github.com/PickNikRobotics/generate_parameter_library/commit/70c917da74608c19203611a7f13532ada89cf8f5#r111546330

Thank you @bijoua29 for calling this out. I'm sorry it took me so long to see your comment. I'll fix this eventually but it is not a priority for me so if you or someone else gets to it first I'd welcome the PR and will try to merge it quickly.

closed 2023-07-17 by tylerjw

bijoua29, 2023-07-18:
Thanks for implementing this

tylerjw, 2023-07-19:
I'm sorry, but it won't be in that sync because I haven't released between merging this and the sync closing. I plan on releasing this library at the end of this week which means it should be in the next sync.

bijoua29, 2023-07-19:
No problem. I just wanted to plan for it. So will wait for next sync and make appropriate changes i.e., call the right overloaded constructor.

PR #132 (2023-07-17): Use node logger
Fixes #125