ros-tooling/generate_parameter_library #129: Error output from subset_of

christophfroehlich, 2023-07-14:
While playing around with the validators for https://github.com/ros-controls/ros2_controllers/pull/703, I found out that the error message of `subset_of` is not really helpful. Launching the JTC from [demo 1](https://github.com/ros-controls/ros2_control_demos/tree/master/example_1) with 
```yaml
    command_interfaces:
      - test
```
gives
> Exception thrown during init stage with message: argument not found 

If I delete this section, I get a useful error from `not_empty`:

> Exception thrown during init stage with message: Invalid value set during initialization for parameter 'command_interfaces': Parameter 'command_interfaces' cannot be empty 

@tylerjw Is there something wrong with the [validator configuration](https://github.com/ros-controls/ros2_controllers/pull/703/files#diff-30f8919bb507925276bf2d4d1232698258244787583f97bb65d1cce950f2cf2bR25) or is this an issue with this library?

tylerjw, 2023-07-17:
Here is the code for subset_of: https://github.com/PickNikRobotics/RSL/blob/91c445b283c5c74a23be0a62f8e6ca07f1fb1e5a/include/rsl/parameter_validators.hpp#L75

You can see the error it should produce. That error seems like you are hitting something else. Can you run that in gdb and get a backtrace to see what is throwing that exception?

christophfroehlich, 2023-07-17:
First time for me to run gdb on that stuff. It seems that `subset_of<>` is returning the correct string, passed to fmt::format 

> 0x7fffc0119570 "Invalid value set during initialization for parameter 'command_interfaces': Entry 'test' in parameter 'command_interfaces' is not in the set {position, velocity, acceleration, effort}"

[image]

christophfroehlich, 2023-07-17:
I tried it with ros2_controllers (rolling and humble) installed from source, I have no binary installation at hand.

tylerjw, 2023-07-19:
I'll need to re-create this locally to determine why our validation error message isn't getting out. I don't understand where `Exception thrown during init stage with message: argument not found` comes from.

closed 2026-03-30 by christophfroehlich