ros-tooling/generate_parameter_library #130: Parameters and gmock

christophfroehlich, 2023-07-14:
I added `not_empty<>` to https://github.com/ros-controls/ros2_controllers/pull/703, but got error messages that the command_interface cannot be empty:
https://github.com/ros-controls/ros2_controllers/actions/runs/5555108925/jobs/10145808720?pr=703

How does this library work together with gmock, and why does the parameter-test fail but the test itself is fine without `not_empty<>`?

tylerjw, 2023-07-17:
Wait, where does this work? 

I think the reason for the errors wit both not_empty and size_gt is that a parameter is not set before you try to initialize, and there is no default, so it defaults to a size of 0 which fails validation. You'll need to select the parameter on the node before creating the ParamListener. You could mock the param listener if you wanted to with gmock if you created a wrapper for it and implemented your own version of the wrapper.

christophfroehlich, 2023-07-17:
Ok, I think I realized now that `subset_of<>` validator, as well as the custom ones, do not throw errors if the parameters are not set.

But actually, they are set before calling the init()/on_init() methods, where the ParamListener is created.

https://github.com/ros-controls/ros2_controllers/blob/8436ed549becce0b694dfe23cfa053436f2b6d42/joint_trajectory_controller/test/test_trajectory_controller_utils.hpp#L157-L174

What is missing to satisfy ParamListener?

christophfroehlich, 2023-07-17:
No, I'm wrong. The fields of params_ will be set, not the ROS parameters used for ParamListener. Why aren't they overridden by ParamListener then?
https://github.com/ros-controls/ros2_controllers/blob/8436ed549becce0b694dfe23cfa053436f2b6d42/joint_trajectory_controller/src/joint_trajectory_controller.cpp#L56-L60

christophfroehlich, 2023-07-17:
And because the node is created inside the init() method of the ControllerInterfaceBase, there is no chance to set the parameters before the ParamListener.

Thanks for pointing me in this direction

closed 2023-07-17 by christophfroehlich