ros-tooling/generate_parameter_library #140: Array of custom sub-structs

christophfroehlich, 2023-08-11:
Is it somehow possible to declare an array of a custom sub-structure? Or how could the library be extended in this way?

Something like

```yaml
cpp_name_space:
  sub_array:
    sub_struct:
      a_parameter: {
        type: double
      }
      another_parameter: {
        type: double
      }
```
and
```yaml
  ros__parameters:
    sub_array:
      - a_parameter: 1.
        another_parameter: 2.0
      - a_parameter: 2.
        another_parameter: 3.0
```

used as `params.sub_array.at(0).a_parameter`

tylerjw, 2023-08-17:
I don't know how that would map into ros parameters themselves. I don't think they have the concept of a map of structures. The current structure implementation is based on the idea of multiple parameters in a namespace and not something more formal like a ros message.

christophfroehlich, 2023-08-18:
you are right, I haven't thought about that end

closed 2023-08-18 by christophfroehlich