ros-tooling/generate_parameter_library #141: ModuleNotFoundError when building with --simlink-install

BrunoB81HK, 2023-08-22:
When building the `example_python` package inside my workspace, I get the `ModuleNotFoundError`. This happens everytime I try to build with `--symlink-install`. I first encountered this bug with my package and I decided to try the example directly to see if it happened with it too and it does. here is the traceback:

```python
Traceback (most recent call last):
  File "/home/user/colcon_ws/install/generate_parameter_module_example/lib/generate_parameter_module_example/test_node", line 33, in <module>
    sys.exit(load_entry_point('generate-parameter-module-example', 'console_scripts', 'test_node')())
  File "/home/user/colcon_ws/install/generate_parameter_module_example/lib/generate_parameter_module_example/test_node", line 25, in importlib_load_entry_point
    return next(matches).load()
  File "/usr/lib/python3.10/importlib/metadata/__init__.py", line 171, in load
    module = import_module(match.group('module'))
  File "/usr/lib/python3.10/importlib/__init__.py", line 126, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1050, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1027, in _find_and_load
  File "<frozen importlib._bootstrap>", line 1006, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 688, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 883, in exec_module
  File "<frozen importlib._bootstrap>", line 241, in _call_with_frames_removed
  File "/home/user/colcon_ws/build/generate_parameter_module_example/generate_parameter_module_example/minimal_publisher.py", line 33, in <module>
    from generate_parameter_module_example.admittance_parameters import (
ModuleNotFoundError: No module named 'generate_parameter_module_example.admittance_parameters'
```

I also checked my `PYTHONPATH` and I think that it may be related to name collisions.

I saw that @pac48 was referencing to some side-effects when using `--symlink-install` in #122, is this related?

BrunoB81HK, 2023-08-22:
I'm using humble with Ubuntu 22.04. We are also using docker and our container is generated from [this one](https://hub.docker.com/layers/library/ros/humble-ros-base/images/sha256-4dfbb2315bf31a21ad04e7fbdf457fc9c849f4599ac76f1cd096bc5faf6d27c1?context=explore).

pac48, 2023-08-22:
Are you building with any extra options? When I run `colcon build --symlink-install` to build and `ros2 run generate_parameter_module_example test_node --ros-args --params-file ros_ws/src/generate_parameter_library/example_python/config/implementation.yaml` to run, it works even on the image you mentioned.

BrunoB81HK, 2023-08-22:
OK, that is weird. The only other that we use is the following `--cmake-args -DCMAKE_BUILD_TYPE=Release`. Maybe the error is on my side.

pac48, 2023-08-22:
Okay, that is strange. The only other thing I'd suggest is to ensure that rosdep is run: `rosdep install --from-paths src --ignore-src -y ` before building and running.

BrunoB81HK, 2023-08-29:
@pac48 is there any way you can share me the `Dockerfile` you used that worked for you?

pac48, 2023-08-30:
@BrunoB81HK this is the [image](https://hub.docker.com/layers/osrf/ros/humble-desktop-full/images/sha256-3a4da32b03616edf8c9cf0d2ac39659ed3ea37f6d3da25209496048141c5cff4?context=explore) that I used. Its called `osrf/ros:humble-desktop-full`

BrunoB81HK, 2023-09-18:
After much investigating, I found out that the probleme lied in my Dockerfile configuration. The build process was not saving any changes made inside the `src/` directory. I'll close this issue since it is not a bug. Thanks a lot for your help!

closed 2023-09-18 by BrunoB81HK

ahi-bplus, 2024-04-15:
@BrunoB81HK can you share the changes you made to fix the problem? I have exactly the same problem with --symlink-install

BrunoB81HK, 2024-04-16:
@ahi-bplus Are you building the project inside your `Dockerfile`? If so, the problem for me was pretty simple. 

When using `--symlink-install`, it creates a new module inside the `src/` directory. The module is accessible because, if I remember correctly, the whole python module directory (the one who contains the `__init__.py` file) is symlinked, not just the files. This is not a problem in itself, but becomes one if you mount your `src/` into the container. If you do so, the module inside the `install/` directory changes to the one mounted in your `src/` directory. For that reason, the parameter module is overwritten.

I can give you some ideas to help with your case:
1. Avoid using `--symlink-install` inside the `Dockerfile`.
2. Build your project inside the docker container, not the `Dockerfile`.
3. If used for development, start using devcontainers.
4. Find a way to avoid mounting your `src/` directory inside the container.

I hope it helps, feel free to ask more questions if you need! :)

ahi-bplus, 2024-04-16:
@BrunoB81HK Thanks for your quick reply. I already use a devcontainer for development. I mount my src files accordingly. As soon as I build my ROS node which implements `generate_parameter_library` using `colcon build --symlink-install --merge-install`, I get the problem with `ModuleNotFoundError`. Without `--symlink-install --merge-install` it works.

ahi-bplus, 2024-04-16:
Okay, I have found the problem. It has nothing to do with `--symlink-install` or `--merge-install`. The problem comes from my package structure.

```bash
package_name/
├── README.md
├── __init__.py
├── config
│   └── default
│       └── default.yaml
├── models
├── package.xml
├── requirements.txt
├── resource
│   └── package_name
├── setup.cfg
├── setup.py
├── src
│   └── package_name
│       ├── __init__.py
│       ├── node.py
│       ├── parameters.yaml
```
Unfortunately the `package_dir={'': 'src'},` of `generate_parameter_library` is not considered. If I avoid the subdirecory `src/package_name` everything works. I think this would be worth an issue?

BrunoB81HK, 2024-04-16:
If you think this is a bug, you should create a new issue. Even if it's not, that will give others that have the same problem some help too.