ros-tooling/generate_parameter_library #142: Mapped parameters can't refer to base class parameters

bijoua29, 2023-08-26:
This is probably not a bug but a feature request. In any case, I had a situation where I had joints parameter defined in a base class. In the derived class, I needed to use a mapped parameter that referred to the base class joints parameters. This would not compile.. I got around it by re-declaring the same joints parameter in the derived class yaml file. This worked but is not ideal since the joints parameter is now declared in two places.

tylerjw, 2023-08-28:
Yea, in some ways, this is a limitation of the parameter system in ROS 2. We could add features for this, but I'm unsure how to implement it. Currently, each generated parameter library is independent of others and there is no way to refer to another parameter library from a second one.

bijoua29, 2023-08-29:
Yeah, I agree there doesn't seem to be an obvious way to implement this. I wonder if just some way of specifying an include path to the base class GPL library include in the yaml file might suffice. Obviously, there are multiple ways of this going wrong though.