ros-tooling/generate_parameter_library #157: ModuleNotFoundError: No module named 'generate_parameter_library_py'

bchoineubility, 2023-11-16:
Hi 
I am trying to build moveit-core with yocto recipe. by the way, I faced configure error at first.
```
| CMake Error at /home/bchoi/work/official_ros_humble_build/bchoi-build/tmp/work/armv8a-poky-linux/moveit-core/2.5.5-1-r0/recipe-sysroot/usr/share/generate_parameter_library/cmake/generate_parameter_library.cmake:33 (message):
|   generate_parameter_library_cpp() variable
|   'generate_parameter_library_cpp_BIN' must not be empty
| Call Stack (most recent call first):
|   online_signal_smoothing/CMakeLists.txt:31 (generate_parameter_library)
```

I copied generate_parameter_lib_cpp python script to sysroot and configure error would be fixed at the time.
But, the new compile error is occurred : 

```
| Traceback (most recent call last):
|   File "/home/bchoi/work/official_ros_humble_build/bchoi-build/tmp/work/armv8a-poky-linux/moveit-core/2.5.5-1-r0/recipe-sysroot/usr/bin/generate_parameter_library_cpp", line 5, in <module>
|     from generate_parameter_library_py.generate_cpp_header import main
| ModuleNotFoundError: No module named 'generate_parameter_library_py'

```

Can you help me ?

BR,
Mark

pac48, 2023-11-24:
It looks like you do not have `generate_parameter_library` installed. The moveit2 `moveit_core` package depends on `generate_parameter_library` in its package.xml. Have you run rosdep to install all dependencies? Maybe try this in your ros2 workspace:  `rosdep install --from-paths src --ignore-src -r -y`

bchoineubility, 2023-11-27:
Hi @pac48 

I will try it thanks.

BR,
Mark

bchoineubility, 2023-11-29:
Hi @pac48 

It works for me. thanks.

closed 2023-11-29 by bchoineubility