ros-tooling/generate_parameter_library #167: Error building python example

wouterio, 2024-01-05:
Hi,

I'm trying to run [your python example](https://github.com/PickNikRobotics/generate_parameter_library/tree/main/example_python) on Ubuntu 22.04 with Ros2 Humble installed.

Following these instructions:
```
 mkdir colcon_ws
 mkdir colcon_ws/src
 cd colcon_ws/src
 git clone https://github.com/picknikrobotics/generate_parameter_library.git
 cd ..
 colcon build
```
gives me this error:
```
oddbot@oddbot-fitlet:~/gen_param_lib_example$ mkdir colcon_ws
oddbot@oddbot-fitlet:~/gen_param_lib_example$ mkdir colcon_ws/src
oddbot@oddbot-fitlet:~/gen_param_lib_example$ cd colcon_ws/src
oddbot@oddbot-fitlet:~/gen_param_lib_example/colcon_ws/src$ git clone https://github.com/picknikrobotics/generate_parameter_library.git
Cloning into 'generate_parameter_library'...
remote: Enumerating objects: 1844, done.
remote: Counting objects: 100% (821/821), done.
remote: Compressing objects: 100% (392/392), done.
remote: Total 1844 (delta 624), reused 541 (delta 426), pack-reused 1023
Receiving objects: 100% (1844/1844), 400.36 KiB | 2.92 MiB/s, done.
Resolving deltas: 100% (1112/1112), done.
oddbot@oddbot-fitlet:~/gen_param_lib_example/colcon_ws/src$ cd ..
oddbot@oddbot-fitlet:~/gen_param_lib_example/colcon_ws$ colcon build
Starting >>> generate_parameter_library_py
Starting >>> parameter_traits
--- stderr: generate_parameter_library_py
/usr/lib/python3/dist-packages/setuptools/command/install.py:34: SetuptoolsDeprecationWarning: setup.py install is deprecated. Use build and pip and other standards-based tools.
  warnings.warn(
---
Finished <<< generate_parameter_library_py [1.97s]
--- stderr: parameter_traits
CMake Error at CMakeLists.txt:8 (find_package):
  By not providing "Findrsl.cmake" in CMAKE_MODULE_PATH this project has
  asked CMake to find a package configuration file provided by "rsl", but
  CMake did not find one.

  Could not find a package configuration file provided by "rsl" with any of
  the following names:

    rslConfig.cmake
    rsl-config.cmake

  Add the installation prefix of "rsl" to CMAKE_PREFIX_PATH or set "rsl_DIR"
  to a directory containing one of the above files.  If "rsl" provides a
  separate development package or SDK, be sure it has been installed.

---
Failed   <<< parameter_traits [2.14s, exited with code 1]

Summary: 1 package finished [3.66s]
  1 package failed: parameter_traits
  2 packages had stderr output: generate_parameter_library_py parameter_traits
  3 packages not processed
oddbot@oddbot-fitlet:~/gen_param_lib_example/colcon_ws$
```

Any ideas why I get this error?

tylerjw, 2024-01-05:
This is probably because you don't have rsl installed. If you run rosdep install on your source directory it should do it for you. Here is an example how you might do that:

```
rosdep install -r --from-paths src --ignore-src --rosdistro humble -y
```

wouterio, 2024-01-10:
Ah ok I'll give it a try. Perhaps good to mention this in the instructions?

wouterio, 2024-01-11:
Ok so tried but got this error:

```
oddbot@oddbot-fitlet:~/Documents/repos/colcon_ws$ rosdep install -r --from-paths src --ignore-src --rosdistro humble -y
executing command [sudo -H apt-get install -y ros-humble-rsl]
[sudo] password for oddbot:
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
The following NEW packages will be installed:
  ros-humble-rsl
0 upgraded, 1 newly installed, 0 to remove and 16 not upgraded.
Need to get 17.1 kB of archives.
After this operation, 89.1 kB of additional disk space will be used.
Ign:1 http://repo.ros2.org/ubuntu/main jammy/main amd64 ros-humble-rsl amd64 1.0.1-1jammy.20231207.184225
Err:1 http://packages.ros.org/ros2/ubuntu jammy/main amd64 ros-humble-rsl amd64 1.0.1-1jammy.20231207.184225
  404  Not Found [IP: 54.176.191.34 80]
E: Failed to fetch http://packages.ros.org/ros2/ubuntu/pool/main/r/ros-humble-rsl/ros-humble-rsl_1.0.1-1jammy.20231207.184225_amd64.deb  404  Not Found [IP: 54.176.191.34 80]
E: Unable to fetch some archives, maybe run apt-get update or try with --fix-missing?
ERROR: the following rosdeps failed to install
  apt: command [sudo -H apt-get install -y ros-humble-rsl] failed
  apt: Failed to detect successful installation of [ros-humble-rsl]
```

pac48, 2024-01-12:
@wouterio You might have to run `sudo apt update` on your machine.

wouterio, 2024-01-15:
Yes thanks, that worked.

closed 2024-01-15 by wouterio