ros-tooling/generate_parameter_library #170: Code generation gives syntax errors with newlines

wouterio, 2024-01-16:
A description containing newlines
```
  hover_override: {
    type: int,
    default_value: 1,
    description: "Override hover action:\n0: Hover\n1: Push\n2: Pull\n-1: Do not override",
    validation: {
      one_of<>: [ [ 0, 1, 2, -1 ] ],
    },
  }
```
leads to generated python code with syntax errors:
```
            if not self.node_.has_parameter(self.prefix_ + "hover_override"):
                descriptor = ParameterDescriptor(description="Override hover action:   <<< missing quote
                0: Hover
                1: Push
                2: Pull
                -1: Do not override", read_only = False)
                parameter = updated_params.hover_override
                self.node_.declare_parameter(self.prefix_ + "hover_override", parameter, descriptor)
```

christophfroehlich, 2024-01-16:
I had a similar issue with generated c++ code and implemented a jinja filter for this, see #164
Maybe you can use it (or implement something similar) for the python templates.

wouterio, 2024-01-23:
I could easily avoid the problem skipping the newlines. But I'd say it's still something to be fixed.

wouterio, 2024-02-27:
Ha that was quick, thank you!