ros-tooling/generate_parameter_library #171: Building from source on Ubuntu 20.04 fails

qleonardolp, 2024-01-19:
I've been working on a environment that needs to run on **Ubuntu 20.04**, with ros2 **Humble**. I'm trying to build `ros2_controllers`, then I realized that it was missing the build of the dependencies such as `generate_parameter_library`. Since in this case we are using a unofficial ubuntu version for Humble, I'm build from source most of the dependencies. Until this moment what is working:

1. Installed `fmt`: apt install libfmt-dev
2. Cloned [PickNikRobotics/cpp_polyfills](https://github.com/PickNikRobotics/cpp_polyfills.git)
3. colcon build --packages-select tcb_span tl_expected
4. Cloned [PickNikRobotics/RSL](https://github.com/PickNikRobotics/RSL.git)
5. Unfortunately, in this one it was necessary to set the cmake_minimum_required to 3.16 at the CMakeList. The build worked, but I don't know how wrong this can be
6. Installed `typeguard`: $pip3 install typeguard
7. Cloned `generate_parameter_library`
8. Builded generate_parameter_library_py, parameter_traits, and generate_parameter_library using colcon
9. The error comes when building generate_parameter_library_example. The building fail with the following message:

(wip)

christophfroehlich, 2024-01-19:
Ubuntu 20.04 is not unoffical for humble, but [Tier 3](https://www.ros.org/reps/rep-2000.html#humble-hawksbill-may-2022-may-2027).

For RSL and CMake version see https://github.com/PickNikRobotics/RSL/issues/113

Maybe this solves your problem: https://github.com/PickNikRobotics/generate_parameter_library/issues/124

qleonardolp, 2024-01-19:
> Ubuntu 20.04 is not unoffical for humble, but [Tier 3](https://www.ros.org/reps/rep-2000.html#humble-hawksbill-may-2022-may-2027).
> 
> For RSL and CMake version see [PickNikRobotics/RSL#113](https://github.com/PickNikRobotics/RSL/issues/113)
> 
> Maybe this solves your problem: #124

I'll check this last issue and their proposed solution. Thanks for your fast reply. I had to stop writting the issue and couldn't copy/paste the erro log from the build. In a few hours I going to update my issue with these datails and any other advance from #124.

qleonardolp, 2024-01-19:
I realized it was just the old cmake version the issue. After [build and install from source cmake 3.28](https://askubuntu.com/questions/355565/how-do-i-install-the-latest-version-of-cmake-from-the-command-line) the the build worked fine, such as building the packages that depends of this, in my case ros2_controllers `joint_state_broadcaster` and `forward_command_controller`

christophfroehlich, 2024-01-19:
I see now that this library has a wrong `cmake_minimum_required` declaration, `cmake_path` is available only from 3.20 on:
https://cmake.org/cmake/help/latest/command/cmake_path.html

https://github.com/PickNikRobotics/generate_parameter_library/blob/f0d9819b0e6875d3e6e6dd66846b76aa72159d50/generate_parameter_library/CMakeLists.txt#L1

christophfroehlich, 2024-02-06:
yes :+1:

closed 2024-02-06 by pac48