ros-tooling/generate_parameter_library #174: Add `# flake8: noqa` tag to generated files

wouterio, 2024-01-23:
We're using the generate parameter library for a project that we build with `colcon build --symlink-install`. With `--symlink-install` the generated python files also show up in our source folder.

That means our code inspectors go through the generated py files as well. And those files don't pass the inspection. It would be nice if the generated py files would have the `# flake8: noqa` tag (and equivalents for other inspectors), so that the inspectors skip these files.

Or even better, it would be nice if the `generate_parameter_module` function gets a parameter `tags: list[str]` that callers can use to specify any of these tags themselves.

Timple, 2024-02-25:
It's interesting that these files end up in your source folder right?

The files should be generated either directly in the `install` folder or in the `build` folder and the moved to the install folder.
Having symlinks shouldn't influence this. We use symlinks and I didn't notice any build artifacts thus far in our `src` folder.

pac48, 2024-02-25:
> symlinks

I believe this happens on ROS humble. I think the entire directory is symlinked, so there is not an easy way around it.  @Timple Are you on a new version of ROS, maybe Iron?

Timple, 2024-02-26:
Ah yes, we're running iron. So that might be different.

wouterio, 2024-02-26:
Yeah we're running Humble here. We now add the noqa tag ourselves during build after generating the py files. I guess it's a minor fix to add it during generation?

pac48, 2024-02-26:
This change has been merged