ros-tooling/generate_parameter_library #179: Question: how to handle an array of custom data structures?

reinzor, 2024-02-29:
Hi,

Thanks for sharing this library! I have a question about using this library with an array of parameters of custom data structures. e.g.

```yaml
plugins:
  - {plugin: pluginA, enabled: true}
  - {plugin: pluginB, enabled: true}
```

As far as I know, this is not supported by the ROS2 parameter system since every parameter need an unique key: so arrays of complex types are not supported.

In the nav2 system, this problem is solved as follows:

https://github.com/ros-planning/navigation2/blob/654e7e0d39c515612713c7426a30b2a219ae477a/nav2_bringup/params/nav2_params.yaml#L191

So we have an additional parameter that specifies the keys of the complex data structures:

```yaml
plugins: ["pluginA", "pluginB"]
pluginA: {} # Complex data structure
```

How should we address this problem when using this library?

Thanks!

christophfroehlich, 2024-02-29:
for reference https://github.com/PickNikRobotics/generate_parameter_library/issues/140

but maybe you have found a workaround?

reinzor, 2024-02-29:
Thanks for the reference. So out of this I can conclude that this is not supported? This could be realized by introducing an additional feature that supports variables in parameter keys. Any thought about this by any of the creators, maintainers?

christophfroehlich, 2024-02-29:
but how does nav2 declare the parameters? or do they use `allow_undeclared_parameters` and parse it based on the data from `plugins`?

reinzor, 2024-02-29:
Every parameter is declared in code. The `plugin` parameter is declared [here](https://github.com/ros-planning/navigation2/blob/bf3afa32287f1dd3673b56e90b26f40fa7cf8e66/nav2_costmap_2d/src/costmap_2d_ros.cpp#L124). Based on the value of this parameter, the other variables are declared [here]. And then the plugin specific parameters are declared [here](https://github.com/ros-planning/navigation2/blob/bf3afa32287f1dd3673b56e90b26f40fa7cf8e66/nav2_costmap_2d/plugins/obstacle_layer.cpp#L79). So the parameter keys depend on the values of the `plugin` parameter.

reinzor, 2024-02-29:
This pattern is also described here: https://github.com/mikeferguson/ros2_cookbook/blob/b7250b3270a8399f5fa2b894472b591f6a6914c8/rclcpp/parameters.md?plain=1#L93

reinzor, 2024-03-19:
That worked.

closed 2024-03-19 by reinzor