ros-tooling/generate_parameter_library #180: Changing Parameter Library Header Installation Location

jcarpinelli-bdai, 2024-03-06:
Is it possible to change the location of installed parameter library header files? For example, in your [example](https://github.com/PickNikRobotics/generate_parameter_library/blob/4b50b985447ac2b20e6b54eb695ea11725a19d89/example/include/generate_parameter_library_example/minimal_publisher.hpp#L34) you include the parameter file header without a folder prefix. Is there any way to change this library's header install directory? I am trying to remove the `<parameter_library_name>/include` prefix, so I can `#include "parent_package/parameter_library_header.hpp`.

Timple, 2024-06-27:
This would help us too! :slightly_smiling_face:

AugusteBourgois, 2024-08-05:
Hi everyone, PR https://github.com/PickNikRobotics/generate_parameter_library/pull/213 should fix the issue. It's on a fork right now, but can hopefully be merged soon. Open to discussion anyway.

closed 2025-01-14 by sea-bass