ros-tooling/generate_parameter_library #186: How does the installation process work?

Guelakais, 2024-03-21:


pac48, 2024-03-29:
@Guelakais Which ROS distro are you using? You should be able to build this the same as other ROS 2 packages. 1) create a colcon workspace 2) clone this into the src folder 3) run rosdep from the base of the workspaces, 4) finally, run colcon build

shu-13, 2024-04-15:
@Guelakais Hi, you might want to use `--packages-up-to` option instead of `--packages-select` in this case.
colcon says that it can't find the dependent package specified in the package.xml of generate_parameter_library
```bash
[0.983s] ERROR:colcon.colcon_cmake.task.cmake.build:Failed to find the following files:
- /ros_ws/install/generate_parameter_library_py/share/generate_parameter_library_py/package.sh
- /ros_ws/install/parameter_traits/share/parameter_traits/package.sh
Check that the following packages have been built:
- generate_parameter_library_py
- parameter_traits
```

Also follow the steps as @pac48 mentions to make sure all the dependent packages are installed.

closed 2024-04-15 by Guelakais