ros-tooling/generate_parameter_library #190: Statically Sized Parameters

jcarpinelli-bdai, 2024-04-01:
Is it possible to configure `std::array`-valued parameters for fixed-size definitions? For example, for Cartesian stiffness gains, I'm currently using...

```yaml
cartesian_impedance_parameters:
  stiffness: 
    x: {
        type: double,
    }, 
    y: {
        type: double,
    },
    ...
```

It may be nice to have a single parameter for stiffness gains, but still have stack-allocated values. For example...

```yaml
cartesian_impedance_parameters:
  stiffness: {
    type: double_array_fixed_06,
  }
```

**Do you agree this feature does not currently exist, but may be a nice feature to have?**

pac48, 2024-04-07:
@jcarpinelli-bdai we do support a feature that is similar to what you are asking for. You can check [here](https://github.com/PickNikRobotics/generate_parameter_library/blob/268d5e439cd6c8948e42b90ff3c16a4ffe2f0cfa/example/src/parameters.yaml#L100) in the example for how it is used. The referenced example creates a vector that is at most size 10 and is stack-allocated. If you want to require a specific size, you could probably use `gt<>` and `lt<>` validators.

To answer your question, we somewhat support that and it is useful. Also, you may have issues with the leading zero in front of the 6. I'm not positive if that is valid.

jcarpinelli-bdai, 2024-04-12:
Thank you, @pac48! To clarify, what value should the resulting parameter take? Is it an `std::vector` that is somehow stack allocated, with a maximum size? Or is it supposed to be an `std::array`?

pac48, 2024-04-12:
@jcarpinelli-bdai The type is a `rsl::StaticVector<double, 10>`, which is a custom type in the Robot Support Library. Here is the definition https://github.com/PickNikRobotics/RSL/blob/e93c5456c2a328de6ae18f4b663a7dbc74c1a81d/include/rsl/static_vector.hpp#L18 . Under the hood, it does use a  `std::array`.

closed 2026-11-13 by christophfroehlich