ros-tooling/generate_parameter_library #191: Allow nested map definitions

Kotochleb, 2024-04-03:
Currently, the library allows only top level map definitions. Something that would be nice is the option to create parameters in a following way.

With this yaml:
```yaml
my_node:
  my_params:
    params_to_be_mapped: {
      type: string_array,
    }
    __map_params_to_mapped:
      in_map_param: {
        type: bool,
      }
```
When attempting to do so, the current Jinja template will create the following code:
```python
# declare any new dynamic parameters
for value_1 in updated_params.params_to_mapped:
  updated_params.my_params.add_entry(value_1)

# ...

for value_1 in updated_params.params_to_mapped:
  param_name = f"{self.prefix_}my_params.{value_1}.in_map_param"

# ...

# declare and set all dynamic parameters
for value_1 in updated_params.params_to_mapped:
  updated_params.my_params.add_entry(value_1)
```

While the expected output is:
```python
# declare any new dynamic parameters
for value_1 in updated_params.my_params.params_to_mapped:
  updated_params.my_params.add_entry(value_1)

# ...

for value_1 in updated_params.my_params.params_to_mapped:
  param_name = f"{self.prefix_}my_params.{value_1}.in_map_param"

# ...

# declare and set all dynamic parameters
for value_1 in updated_params.my_params.params_to_mapped:
  updated_params.my_params.add_entry(value_1)
```

Namespace of `my_params` is not taken into account in `for` loops of the mapped values.

Topic initially started in https://github.com/PickNikRobotics/generate_parameter_library/pull/183.

BrunoB81HK, 2024-04-18:
Is there any plan to implement this feature? This would be very useful since a mapped parameters is not necessarily a global one.

@pac48 suggested to use the following key format and I think this is the way to go: `__map_my_params.params_to_mapped`

pac48, 2024-04-18:
@BrunoB81HK I am working on a branch here  https://github.com/PickNikRobotics/generate_parameter_library/tree/pr-allow-mapped-sub-parameter but it is not ready yet.

BrunoB81HK, 2024-04-18:
@pac48 Great! Thanks a lot for your work!

YannickdeHoop, 2025-03-19:
@BrunoB81HK, what is the progress of your branch?

pac48, 2025-03-19:
The feature was merged in this PR https://github.com/PickNikRobotics/generate_parameter_library/pull/166

closed 2025-03-19 by pac48