ros-tooling/generate_parameter_library #204: How to include the parameters in a dependent package?

reinzor, 2024-05-30:
Hi,

Thanks for this library!

We would like to include a public header from another packages that includes the parameter header. However, this header is not available for that package as `#include <example_parameters.hpp>` but it is installed under `example_parameters/example_parameters.hpp`. How can we do this?

So the setup:

package_with_parameters/include/header.hpp

```
#include <example_parameters.hpp>
```

another_package/include/header.hpp

```
#include <package_with_parameters/header.hpp> // #include <example_parameters.hpp> cannot be found, example_parameters/example_parameters.hpp can be found
```

Can you help us with this?

christophfroehlich, 2024-05-30:
I found a way to install the created headers
```CMake
install(FILES ${CMAKE_CURRENT_BINARY_DIR}/limiters_parameters/include/limiters_parameters.hpp
  DESTINATION include/${PROJECT_NAME}/${PROJECT_NAME})
```
and now another package can 

```c++
#include "steering_controllers_library/limiters_parameters.hpp"
```

But there is for sure a smarter way. Maybe someone with knowledge in CMake can enlighten us ;)

reinzor, 2024-05-31:
Thanks for your quick response.

Including the parameter file via another package already works. The problem here is that the include is different for the package that holds defines the parameters and a dependent package. This include:

```cpp
#include "steering_controllers_library/limiters_parameters.hpp"
```

is not being resolved correctly by the package itself. There, it is available as

```cpp
#include "limiters_parameters.hpp"
```

Or am I missing something?

christophfroehlich, 2024-05-31:
That's how I use it now.
but :eyes: https://github.com/PickNikRobotics/generate_parameter_library/issues/180

reinzor, 2024-05-31:
Thanks! This would indeed allow for including a header that includes the parameter header in another package.

AugusteBourgois, 2024-08-05:
Hi everyone, PR #213 should fix the issue. It's on a fork right now, but can hopefully be merged soon. Open to discussion anyway.

closed 2025-01-14 by sea-bass