ros-tooling/generate_parameter_library #206: Store codegen result for python code in build folder

Flova, 2024-06-14:
Building a `setup.py` based setup results in code being generated in the source folder.

A cleaner approach would be probably to store this not in the `src` directory. This would also be more consistent with the way the cpp package is build.

Are there any non-obvious issues preventing this behavior?

pac48, 2024-11-15:
Are you building with colcon and symlink-install? In that case, the generated code is symlinked to the src directory by colcon and I’m not sure if there is a way around that.

christophfroehlich, 2026-01-30:
Cross-reference: https://github.com/PickNikRobotics/generate_parameter_library/issues/233#issuecomment-3826224423

christophfroehlich, 2026-04-18:
I vibe-coded a possible fix for this in #348. @Flova could you have a look, test it, and leave a review if the GPT-generated code makes sense?

christophfroehlich, 2026-11-13:
Summary of where this stands:

- #393 (merged) makes regular builds of `setup.py`-based packages generate the parameter module into the build tree. It is installed from there, so nothing is written into `src/`.
- `colcon build --symlink-install` and `pip install -e` still generate into the source package. The generated module has to be part of the user's package, and in those modes the source tree is what Python imports. This is the non-obvious issue @pac48 pointed out above.
- #348 was an earlier attempt and was replaced by #393.

NoDL's Python generator solves the remaining case by generating into a separate namespace package that is installed into the install space, even for symlink installs. I opened #402 with a proposal to do the same here, since that changes the import path and needs its own discussion. Closing this one in favour of #402.

closed 2026-11-13 by christophfroehlich