ros-tooling/generate_parameter_library #217: Including Other Parameters in Validator Inputs

jcarpinelli-bdai, 2024-08-21:
Is there any way to use other parameter values as inputs to custom validators? For example, say we have URDF and SRDF parameters, and we want to make sure they are consistent with one another in a custom validator. 
```
robot_description_parameters:
  urdf: 
    type: string
  srdf:
    type: string
    validation:
      "robot_description_validators::srdf_compatible": [urdf]
```

pac48, 2024-08-30:
This would be a nice feature to have. It is not currently supported. I'd welcome a contribution if anyone has the time. I can not look into it in the short term. This would require changes to the code generator parser, syntax checking, and validation calls.

JGyimah, 2026-10-12:
I have an implementation of this on a [(branch)](https://github.com/JGyimah/generate_parameter_library/tree/feature/cross-parameter-constraints) and would like a sanity check on the design before opening a PR, because what I built is a counter-proposal rather than the syntax suggested above.

### Why not parameter references inside `validation:`

The syntax in the issue body hangs the check off one parameter and passes the other as an argument:

```yaml
srdf:
  type: string
  validation:
    "robot_description_validators::srdf_compatible": [urdf]
```

Two things pushed me away from this:

1. **`urdf` is ambiguous with the string literal `"urdf"`.** Validator arguments today are literals, and `srdf`/`urdf` are `string` parameters, so in this exact example there is no way to tell a reference from a value. Any disambiguation rule would have to be a sigil or a wrapper, at which point the argument list is no longer "just literals".
2. **The constraint is not a property of either parameter.** Attaching it to `srdf` is arbitrary — it reads `urdf` just as much — and it leaves the generator with nowhere sensible to document the relationship, since each parameter's generated section describes only itself.

### What I implemented instead

A reserved `__constraints` key holding named constraints over the parameters of the struct it appears in, at any nesting level:

```yaml
to_base:
  vel_min:
    type: double
    default_value: 0.5
  vel_max:
    type: double
    default_value: 1.6

  __constraints:
    velocity_ordering:
      lt: [$vel_min, $vel_max]
      description: "vel_min must be strictly less than vel_max"
```

Operands are a `$reference` to a declared parameter or a scalar literal; the sigil removes the ambiguity above in both directions. References resolve relative to the enclosing struct, falling back to the file root.

There are three levels, because the cases in this issue and in #301 are not the same shape:

**Relations** — `lt`, `gt`, `lt_eq`, `gt_eq`, `eq`, `ne`, reusing the existing validator vocabulary and emitted as inline comparisons, so no new RSL function is needed and C++ and Python behave identically.

**Conditional** — a requirement that applies only while another parameter holds a value. This covers #301 (the allowed framerate depending on whether the sensor is at full resolution) and mode-dependent limits generally:

```yaml
    manual_mode_speed_limit:
      when:
        eq: [$navigation_mode, "manual"]
      require:
        lt_eq: [$vel_max, 2.0]
```

**Custom functions** — the case in this issue, for predicates that need parsing or iteration:

```yaml
    srdf_matches_urdf:
      "robot_description_validators::srdf_matches_urdf": [$urdf, $srdf]
```

```cpp
tl::expected<void, std::string> srdf_matches_urdf(
    std::string const& urdf, std::string const& srdf);
```

The referenced values are passed positionally rather than as a templated `Params const&`. The user validation header is included above `struct Params` in the generated header, so a non-template signature cannot name it anyway; passing values also keeps the signature checked by the compiler rather than at instantiation, and leaves the generator knowing which parameters each constraint reads.

### Where they are enforced

After every per-parameter validator has passed, at two points:

* `declare_params()`, once defaults, YAML and command-line overrides are merged — throws `InvalidParameterValueException`, so the node fails to construct.
* the update callback, against the prospective state, before `update_internal_params()`. A violation returns an unsuccessful `SetParametersResult`, nothing is committed, and the user callback does not fire.

The caveat worth knowing is that enforcement is only as atomic as the triggering call: moving a coupled pair to a disjoint range needs `set_parameters_atomically`, since `set_parameters` and `ros2 param set` apply values one at a time. That is inherent to the parameter API rather than to this design, and it is documented.

### Checks at generation time

Because the relations are declarative rather than opaque C++, several mistakes are caught before a node ever runs: unknown or mistyped references, references into a `__map_` block (not supported), wrong operand types, a missing validation header for a custom function, and defaults that already violate a constraint:

```
ERROR: Constraint 'velocity_ordering' is violated by the declared default values: 2.0 < 1.6 is false.
```

A `when` guard is also checked against a `one_of` validator on the parameter it references, so a guard comparing against a value outside the allowed set is reported instead of silently becoming dead code. Requirements under a guard are only folded when the guard is active at the defaults.

### Questions before I open a PR

1. Is a separate `__constraints` block acceptable, or would you rather see this inside `validation:` with some other disambiguation?
2. Is `$` the right sigil, or do you prefer something else?
3. Constraints cannot reference `__map_` parameters. Declaring them inside a mapped scope and evaluating per key is a plausible follow-up — is that wanted in the first pass?

Happy to adjust any of this; I would rather settle the syntax here than in review.