ros-tooling/generate_parameter_library #235: How to support multiple nodes in one package

kjeremy, 2025-01-09:
If I have two nodes `A` and `B` and they have different parameter sets but with some overlap do I need to have two separate yaml files and thus two `generate_parameter_library` calls in my `CMakeLists.txt`?

christophfroehlich, 2025-01-09:
If you don't care that you add the additional unused parameters for the other node respectively, you can just generate one parameter library and link it to both executables.

Otherwise, you need to generate two libraries but can't share the common parameters. To do so, you need three libraries (node A, node B, common) and link the common one to both nodes.

minku1219, 2025-10-16:
@christophfroehlich, Kindly reflect your thoughts on this.

christophfroehlich, 2025-10-16:
Would be a nice addition, but is currently not possible.
You will have to extend the [cmake function](https://github.com/PickNikRobotics/generate_parameter_library/blob/main/generate_parameter_library/cmake/generate_parameter_library.cmake), but I don't know how cmake can support variadic functions.
Additionally, the root element defines the c++ namespace, how would you handle that? Is the goal to have all parameters in a single struct, or just compile the three libs to three different files but in a single cmake library?