ros-tooling/generate_parameter_library #239: Make custom_validators accessible from external library

christophfroehlich, 2025-01-22:
With #213 it is straightforward to use a complete cpp library within another package. But how can I use a custom_validator file from a different package?

Is there a smarter way to include the [file from here](https://github.com/ros-controls/control_toolbox/blob/6611872dbfcb458c4eed192f0aed0566f0b3cdd3/CMakeLists.txt#L86-L90) to another library than
```cmake
find_package(control_toolbox)
# get include dirs from control_toolbox for the custom validators
get_target_property(TB_INCLUDE_DIRS control_toolbox::rate_limiter_parameters INTERFACE_INCLUDE_DIRECTORIES)
generate_parameter_library(diff_drive_controller_parameters
  src/diff_drive_controller_parameter.yaml
  ${TB_INCLUDE_DIRS}/control_toolbox/custom_validators.hpp
)
```
@AugusteBourgois any suggestions?
Changing the macro here to avoid a COPY but just use the correct include path `control_toolbox/custom_validators.hpp`?
https://github.com/PickNikRobotics/generate_parameter_library/blob/891c29196ea3c070d818284b0ea2112ac8520037/generate_parameter_library/cmake/generate_parameter_library.cmake#L42-L54

pac48, 2025-02-03:
@christophfroehlich I think inserting the include directly as specified, e.g. `<control_toolbox/custom_validators.hpp>` makes the most sense. That way, you can call `target_include_directories` or `target_link_libraries` on the generated target to make sure your custom header is in the included paths. For now, maybe the file should still be copied if the relative path exists, but otherwise, just insert it in the generated code anyway.

AugusteBourgois, 2025-02-04:
@christophfroehlich to be sure I understand you problem correctly :
- You have two packages, A and B.
- You want to generate the parameter files in A, i.e. your `.yaml` config file is located there, and you call `generate_parameter_library` from A's CMakeLists.txt
- You have a custom validator, i.e. a `.hpp` file located in B, which you want to use in A's `.yaml` file.

Am I correct ?

christophfroehlich, 2025-02-04:
exactly 🤔

AugusteBourgois, 2025-02-04:
@pac48 do you mean you could write in A's `CMakeLists.txt` :

```cmake
find_package(B REQUIRED)

generate_parameter_library(my_parameters_target
src/my_parameters_definition.yaml
<B/custom_validator.hpp>
)
target_link_libraries(my_parameters_target B::custom_validator)
```