ros-tooling/generate_parameter_library #256: gt<> validator sets incorrect from_value in parameter descriptor

ottojo, 2025-04-18:
The ranges in the parameter descriptor are exclusive (https://github.com/ros2/rcl_interfaces/blob/rolling/rcl_interfaces/msg/FloatingPointRange.msg#L3), but currently the `gt<>` validator sets the lower limit to the specified value (instead of the next larger value):

https://github.com/PickNikRobotics/generate_parameter_library/blob/13a1043416b62393d284f067c0c52d4db547a175/generate_parameter_library_py/generate_parameter_library_py/jinja_templates/cpp/declare_parameter#L17

This results in inconsistent behavior between parameter descriptor and parameter validation:

```console
jonas@ade $ ros2 param describe /example_node background.gt_zero_float                            
Parameter name: background.gt_zero_float
  Type: double
  Description: Parameter with default value 1.0 and gt<>: 0.0 validation
  Constraints:
    Min value: 0.0
    Max value: 1.7976931348623157e+308
jonas@ade $ ros2 param set /example_node background.gt_zero_float 0.0 
Setting parameter failed: Parameter 'background.gt_zero_float' with the value '0' must be greater than '0'
```

christophfroehlich, 2026-04-07:
With #336 I had a closer look. The validators called from `ros2 param set` always were fine because the use directly the validators from rsl.
As you mentioned, the `ros2 param describe` from `rcl_interfaces::msg::ParameterDescriptor` is not aware of exclusive bounds.
#339 could fix this.

christophfroehlich, 2026-04-17:
@ottojo do you mind having a look at the PR and leave a review there?

closed 2026-05-16 by christophfroehlich