ros-tooling/generate_parameter_library #258: has invalid type: expected [double_array] got [double]

amoyalang, 2025-04-21:
when I create a double parameter,it's always expect double_array.

My yaml file is bellow:
```yaml
yx_robot:
  # This is the manipulator configuration
  manipulator:
    arm_group_name:
      type: string
      default_value: "yx_robot_arm" 
      validation:
        not_empty<>: []
    hand_group_name:
      type: string
      default_value: "yx_robot_hand" 
      validation:
        not_empty<>: []
    eef_name:
      type: string
      default_value: "yx_robot_hand" 
      validation:
        not_empty<>: []
    hand_frame:
      type: string
      default_value: "hand_link" 
      validation:
        not_empty<>: []
    cartesian_ik_frame:
      type: string
      default_value: "Link6"
      validation:
        not_empty<>: []
    grasp_ik_frame:
      type: string
      default_value: "Link6"
      validation:
        not_empty<>: []
    grasp_frame_transform:
      type: double_array
      default_value: [0.0,0.0,0.0,0.0,0.0,0.0]
      validation:
        fixed_size<>: [6]
    approach_object_min_dist:
      type: double
      default_value: 0.05
      validation:
        not_empty<>: 
    approach_object_max_dist:
      type: double
      default_value: 0.15
      validation:
        not_empty<>: 
    lift_object_min_dist:
      type: double
      default_value: 0.05
      validation:
        not_empty<>: 
    lift_object_max_dist:
      type: double
      default_value: 0.15
      validation:
        not_empty<>: 

  gripper: 
    max_position: # 夹爪自身最大行程
      type: double
      default_value: 0.030 
      validation:
        not_empty<>: []
    min_position: # 夹爪自身最小行程
      type: double
      default_value:  0.00
    full_open_position: # 增加夹指之后最大行程
      type: double
      default_value: 0.030 
    full_close_position: # 增加夹指之后最小行程
      type: double
      default_value: 0.010 
    grasping_position:
      type: double
      default_value: 0.020 #合适的夹取行程
```

Then I look at compile file parameter.hpp:
```cpp
struct Params {
        struct Manipulator {
            std::string arm_group_name = "yx_robot_arm";
            std::string hand_group_name = "yx_robot_hand";
            std::string eef_name = "yx_robot_hand";
            std::string hand_frame = "hand_link";
            std::string cartesian_ik_frame = "Link6";
            std::string grasp_ik_frame = "Link6";
            std::vector<double> grasp_frame_transform = {0.0, 0.0, 0.0, 0.0, 0.0, 0.0};
            double approach_object_min_dist = 0.05;
            double approach_object_max_dist = 0.15;
            double lift_object_min_dist = 0.05;
            double lift_object_max_dist = 0.15;
        } manipulator;
        struct Gripper {
            double max_position = 0.03;
            double min_position = 0.0;
            double full_open_position = 0.03;
            double full_close_position = 0.01;
            double grasping_position = 0.02;
        } gripper;
        // for detecting if the parameter struct has been updated
        rclcpp::Time __stamp;
    };
```
It's looks ok,but when I use it 👍 
```bash
[yx_robot_control_node-1] [DEBUG] [1745225220.014621954] [mtc_warpper]: manipulator.arm_group_name: string = yx_robot_arm
[yx_robot_control_node-1] [DEBUG] [1745225220.014739223] [mtc_warpper]: manipulator.hand_group_name: string = yx_robot_hand
[yx_robot_control_node-1] [DEBUG] [1745225220.014747771] [mtc_warpper]: manipulator.eef_name: string = yx_robot_hand
[yx_robot_control_node-1] [DEBUG] [1745225220.014753069] [mtc_warpper]: manipulator.hand_frame: string = hand_link
[yx_robot_control_node-1] [DEBUG] [1745225220.014757762] [mtc_warpper]: manipulator.cartesian_ik_frame: string = Link6
[yx_robot_control_node-1] [DEBUG] [1745225220.014762380] [mtc_warpper]: manipulator.grasp_ik_frame: string = Link6
[yx_robot_control_node-1] [DEBUG] [1745225220.014779245] [mtc_warpper]: manipulator.grasp_frame_transform: double_array = [0, 0, 0, 0, 0, 0]
[yx_robot_control_node-1] [DEBUG] [1745225220.014789394] [mtc_warpper]: manipulator.approach_object_min_dist: double = 0.050000
[yx_robot_control_node-1] terminate called after throwing an instance of 'rclcpp::exceptions::InvalidParameterTypeException'
[yx_robot_control_node-1]   what():  parameter 'manipulator.approach_object_min_dist' has invalid type: expected [double_array] got [double]
[ERROR] [yx_robot_control_node-1]: process has died [pid 52891, exit code -6, cmd '/home/alang/ros2_ws/install/yx_robot_control/lib/yx_robot_control/yx_robot_control_node --ros-args --params-file /tmp/launch_params_n_qzkidg --params-file /tmp/launch_params_kena0njv'].

```
Why?

JafarAbdi, 2025-04-21:
Can share a reproducible example please?

amoyalang, 2025-04-21:
> Can share a reproducible example please?

Hello,my project is a ros2 package.It's path tree is like this:
```bash
.
├── CMakeLists.txt
├── config
│   ├── yx_robot_config_ros2_parameters.yaml
├── package.xml
├── readme.md
```
I generate parameters.hpp with CmakeLists.txt
```cmake
cmake_minimum_required(VERSION 3.8)
project(yx_robot_config)
set(CMAKE_EXPORT_COMPILE_COMMANDS ON)

if(CMAKE_COMPILER_IS_GNUCXX OR CMAKE_CXX_COMPILER_ID MATCHES "Clang")
  add_compile_options(-Wall -Wextra -Wpedantic)
endif()

find_package(ament_cmake REQUIRED)
find_package(ament_index_cpp REQUIRED)
find_package(yaml-cpp REQUIRED)
find_package(generate_parameter_library REQUIRED)

# generate parameters.hpp
generate_parameter_library(parameters config/yx_robot_config_ros2_parameters.yaml)

file(GLOB SRC_FILES "src/*.cpp")

include_directories(
  include
  ${ament_index_cpp_INCLUDE_DIRS}
  ${YAML_CPP_INCLUDE_DIR}
)

add_library(yx_robot_config SHARED ${SRC_FILES})

target_link_libraries(yx_robot_config
  parameters
  yaml-cpp::yaml-cpp
  ament_index_cpp::ament_index_cpp
)

ament_target_dependencies(yx_robot_config
  yaml-cpp
  ament_index_cpp
  generate_parameter_library
)

install(
  TARGETS yx_robot_config parameters
  EXPORT yx_robot_configTargets
  LIBRARY DESTINATION lib
  ARCHIVE DESTINATION lib
  RUNTIME DESTINATION bin
)

ament_export_targets(yx_robot_configTargets HAS_LIBRARY_TARGET)

install(
  DIRECTORY config
  DESTINATION share/${PROJECT_NAME}
)

install(
  DIRECTORY include/
  DESTINATION include/${PROJECT_NAME}
)

ament_package()
```
After colcon build,I view the ros2_ws/install/yx_robot_config/include/parameters.hpp,I can see:
```cpp
struct Params {
        struct Manipulator {
            std::string arm_group_name = "yx_robot_arm";
            std::string hand_group_name = "yx_robot_hand";
            std::string eef_name = "yx_robot_hand";
            std::string hand_frame = "hand_link";
            std::string cartesian_ik_frame = "Link6";
            std::string grasp_ik_frame = "Link6";
            std::vector<double> grasp_frame_transform = {0.0, 0.0, 0.0, 0.0, 0.0, 0.0};
            double approach_object_min_dist = 0.05;
            double approach_object_max_dist = 0.15;
            double lift_object_min_dist = 0.05;
            double lift_object_max_dist = 0.15;
        } manipulator;
        struct Gripper {
            double max_position = 0.03;
            double min_position = 0.0;
            double full_open_position = 0.03;
            double full_close_position = 0.01;
            double grasping_position = 0.02;
        } gripper;
        // for detecting if the parameter struct has been updated
        rclcpp::Time __stamp;
    };
```
It's seems right.
Then I initial param_listener and params in other packages,it is likes this:
```
 param_listener_ = std::make_shared<yx_robot::ParamListener>(node_);
 params_ = std::make_shared<yx_robot::Params>(param_listener_->get_params());
```
You can see the log,my program destoryed when I initial param_listener and params.The log is:
```bash
[yx_robot_control_node-1] [DEBUG] [1745225220.014621954] [mtc_warpper]: manipulator.arm_group_name: string = yx_robot_arm
[yx_robot_control_node-1] [DEBUG] [1745225220.014739223] [mtc_warpper]: manipulator.hand_group_name: string = yx_robot_hand
[yx_robot_control_node-1] [DEBUG] [1745225220.014747771] [mtc_warpper]: manipulator.eef_name: string = yx_robot_hand
[yx_robot_control_node-1] [DEBUG] [1745225220.014753069] [mtc_warpper]: manipulator.hand_frame: string = hand_link
[yx_robot_control_node-1] [DEBUG] [1745225220.014757762] [mtc_warpper]: manipulator.cartesian_ik_frame: string = Link6
[yx_robot_control_node-1] [DEBUG] [1745225220.014762380] [mtc_warpper]: manipulator.grasp_ik_frame: string = Link6
[yx_robot_control_node-1] [DEBUG] [1745225220.014779245] [mtc_warpper]: manipulator.grasp_frame_transform: double_array = [0, 0, 0, 0, 0, 0]
[yx_robot_control_node-1] [DEBUG] [1745225220.014789394] [mtc_warpper]: manipulator.approach_object_min_dist: double = 0.050000
[yx_robot_control_node-1] terminate called after throwing an instance of 'rclcpp::exceptions::InvalidParameterTypeException'
[yx_robot_control_node-1]   what():  parameter 'manipulator.approach_object_min_dist' has invalid type: expected [double_array] got [double]
[ERROR] [yx_robot_control_node-1]: process has died [pid 52891, exit code -6, cmd '/home/alang/ros2_ws/install/yx_robot_control/lib/yx_robot_control/yx_robot_control_node --ros-args --params-file /tmp/launch_params_n_qzkidg --params-file /tmp/launch_params_kena0njv'].

```
It is always think approach_object_min_dist is a double_array.

JafarAbdi, 2025-04-21:
Sorry I can't reproduce it, can you push the full example to github? Thanks!

amoyalang, 2025-04-21:
> Sorry I can't reproduce it, can you push the full example to github? Thanks!

OK,I will do it when I idle.

christophfroehlich, 2026-11-13:
No more replies, closing

closed 2026-11-13 by christophfroehlich