ros-tooling/generate_parameter_library #259: Any reason why examples use a timer instead of ParameterEventHandler?

danielcranston, 2025-04-24:
Seems like a lost opportunity to showcase best practices. Using a `ParameterEventHandler` ([C++](https://docs.ros.org/en/rolling/Tutorials/Intermediate/Monitoring-For-Parameter-Changes-CPP.html)/[Python](https://docs.ros.org/en/rolling/Tutorials/Intermediate/Monitoring-For-Parameter-Changes-Python.html)) seems like a nicer fit when checking for updates instead of using a timer that polls for changes every x seconds.

I'd be happy to make a PR, just want check in case there's a reason behind it I'm not aware of.

amarburg, 2025-05-03:
+1 It would be good to have a canonical code sample (or example) for receiving a callback on a parameter change _and_ using this package (for the validation, etc).

Best I have so far is something like 

```
...
param_listener_ =  std::make_shared<ParamListener>(get_node_parameters_interface());
callback_handle_ = this->add_on_set_parameters_callback(std::bind(
        &MyNode::parametersCallback, this, std::placeholders::_1));
...

rcl_interfaces::msg::SetParametersResult parametersCallback(
      const std::vector<rclcpp::Parameter> &parameters) {

    param_listener_->update(parameters);
    auto params = param_listener_->get_params();

    // use params
}
```

This is functional,  but doesn't sit right, as the listener is update()'ed twice, once here and once in the internal callback within the ParamListener?

christophfroehlich, 2026-11-13:
Duplicate of #109

closed 2026-11-13 by christophfroehlich